jrsteensen/OpenHornet-Software #85: [BUG]: Mag Switches don't trigger as expected

jrsteensen, 2024-03-09:
[Link to dcs-skunkworks issue](https://github.com/DCS-Skunkworks/dcs-bios/issues/134#issuecomment-1986393967)

Current workaround: Use HID to trigger the switch in game.

Arribe, 2024-03-09:
The other work around is to have logic using DCSBios reads to determine when to turn on and off the mag.

Ben-F111, 2024-07-21:
https://github.com/DCS-Skunkworks/dcs-bios/pull/856 for LDT/R "Mag SW" Output, also have Standard codes working for 

APU (Thanks to HOWLINGAPE)
CRANK  (Thanks to HOWLINGAPE)
DUMP 
CANOPY

Ben-F111, 2024-07-21:
> The other work around is to have logic using DCSBios reads to determine when to turn on and off the mag.

Yes, we look to have now solved this