jrsteensen/OpenHornet-Software #89: [Bug]: DCSBios Bug with SwitchMultiPosition call.

Arribe, 2024-03-13:
### Discord Username

Arribe

### Bug Summary

The built in function for SwitchMultiPosition call doesn't work properly when all of the positions are not connected to an Arduino pin. Due to not having debounce logic related to the PIN_NC logic.

### Expected Results

When rotating the multi-position switches the output should reflect the position of the switch.

### Actual Results

The output bounces betwen the pointed to position and the unconnected/default position identified by PIN_NC

### Applicable File Names

Multiple - Sensor Panel is one.

### Release Version

v0.2.0

### Category

DCS-BIOS

### Applicable End Item(s)

Right Console (RCON)

### Hardware is built to print & this software is tested on that hardware?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations and tested the software on it.
- [X] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

### Miscellaneous Info

https://github.com/DCS-Skunkworks/dcs-bios-arduino-library/pull/56

kbastronomics, 2024-03-14:
opened issue with DCS-BIOS
https://github.com/DCS-Skunkworks/dcs-bios-arduino-library/issues/57

jrsteensen, 2024-03-14:
> opened issue with DCS-BIOS [DCS-Skunkworks/dcs-bios-arduino-library#57](https://github.com/DCS-Skunkworks/dcs-bios-arduino-library/issues/57)

Consider submitting a PR to them to help them speed it along? 😄

kbastronomics, 2025-06-15:
i would suggest instead of fixing how DCS-BIOS works which may/may not be possible in this case instead we create a PCB for the rotary switches that would allow them to be used with a single ADC port. and then use the AnalogMultiPos switch

this take 1 pin and how many steps between switch positions
such as
DcsBios::AnalogMultiPos lstNflrSw("LST_NFLR_SW", A0 ,550 );
which would say for every 550 counts of the ADC move the switch 1 position.

I've designed a test PCB to do this.   which should work for the majority if our Rotary switches
[image]