jrsteensen/OpenHornet-Software #102: [Bug]: CI Pipeline Failing on missing SPI.h File

Arribe, 2024-04-02:
### Discord Username

Arribe

### Bug Summary

PR #101 is failing with entry: /home/runner/work/OpenHornet-Software/OpenHornet-Software/libraries/Arduino-FOC-drivers/src/encoders/mt6835/./MT6835.h:8:10: fatal error: SPI.h: No such file or directory    
8 | #include "SPI.h"
      |          ^~~~~~~
compilation terminated.

### Expected Results

The sketch should compile as SPI.h is a base library included with all Arduino platforms.

### Actual Results

The CI fails to complete with the error the SPI.h file is missing.  I even tried including it at the top of the include list on the main sketch.

### Screenshots/Images/Files

[image]

### Applicable File Names

PR 4A9A1-THROTTLE_CONTROLLER.ino #101

### Release Version

v0.3.0

### Category

Libraries & Dependencies (Arduino Libraries, etc)

### Applicable End Item(s)

Throttle

### Hardware is built to print & this software is tested on that hardware?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations and tested the software on it.
- [X] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-04-04:
Have we tried installing the library as a submodule and calling it that way?

Arribe, 2024-04-04:
It's a submodule header call from SimpleFOC...I didn't want to cause problems by adding a base Arduino header as a submodule...but I can certainly try it to see if that works.

Arribe, 2024-04-04:
No luck.  Adding the core ArduinoCore-avr as a sub module didn't work...and probably caused more problems since that is a base library that's included in every Arduino IDE.  Adding SPI.h and SPI.cpp didn't work and caused more header files to be marked as missing.

Arribe, 2024-04-07:
As far as I can tell ESP32 CI is broken.  Trying a test sketch consisting of basically an empty file except including Wire.h or SPI.h CI fails.  Though based on what I can see the following fragments should lead the CI to reaching those files:

OpenHornet Software's include/esp.mk has this line:
ESP_ROOT = $(ROOTDIR)/libraries/arduino-esp32

[makeEspArduino](https://github.com/plerup/makeEspArduino/tree/4171de2874405408da863e33a78d49aa0fd5deef)
/makeEspArduino.mk has the following line:
ESP_LIBS = $(ESP_ROOT)/libraries

The following the OpenHornet Software libraries folder linkes to the github entry below for ESP_ROOT/libraries:
https://github.com/espressif/arduino-esp32/tree/4b3aebe11fa5acc0f8b259105b4d1990b95c399b/libraries

The libraries folder has SPI, Wire directories.